gazebosim/gz-transport #138: NodeShared.cc:96:48: error: no matching function for call to ‘zmq::message_t::message_t(const char*, std: :__cxx11::basic_string<char>::size_type)’

vmatare, 2020-05-15:
When compiling the 8.0.0 release, I get:
```
/var/tmp/portage/net-libs/ignition-transport-8.0.0/work/ignition-transport8_8.0.0/src/NodeShared.cc: In function ‘int sendHelper(zmq::socket_t&, const string&, int)’:                         
/var/tmp/portage/net-libs/ignition-transport-8.0.0/work/ignition-transport8_8.0.0/src/NodeShared.cc:96:48: error: no matching function for call to ‘zmq::message_t::message_t(const char*, std:
:__cxx11::basic_string<char>::size_type)’                                                                                                                                                      
   96 |   zmq::message_t msg(_data.data(), _data.size());                                                                                                                                      
      |                                                ^                                                                                                                                       
In file included from /var/tmp/portage/net-libs/ignition-transport-8.0.0/work/ignition-transport8_8.0.0/src/NodeShared.cc:21:                                                                  
/usr/include/zmq.hpp:258:9: note: candidate: ‘zmq::message_t::message_t(const zmq::message_t&)’                                                                                                
  258 |         message_t (const message_t&);                                                                                                                                                  
      |         ^~~~~~~~~                                                                                                                                                                      
/usr/include/zmq.hpp:258:9: note:   candidate expects 1 argument, 2 provided                                                                                                                   
/usr/include/zmq.hpp:165:16: note: candidate: ‘zmq::message_t::message_t(zmq::message_t&&)’                                                                                                    
  165 |         inline message_t (message_t &&rhs) : msg (rhs.msg)                                                                                                                             
      |                ^~~~~~~~~                                                                                                                                                               
/usr/include/zmq.hpp:165:16: note:   candidate expects 1 argument, 2 provided                                                                                                                  
/usr/include/zmq.hpp:156:16: note: candidate: ‘zmq::message_t::message_t(void*, size_t, void (*)(void*, void*), void*)’                                                                        
  156 |         inline message_t (void *data_, size_t size_, free_fn *ffn_,                                                                                                                    
      |                ^~~~~~~~~                                                                                                                                                               
/usr/include/zmq.hpp:156:16: note:   candidate expects 4 arguments, 2 provided                                                                                                                 
/usr/include/zmq.hpp:149:25: note: candidate: ‘zmq::message_t::message_t(size_t)’              
  149 |         inline explicit message_t (size_t size_)                                       
      |                         ^~~~~~~~~                                                                                                                                                      
/usr/include/zmq.hpp:149:25: note:   candidate expects 1 argument, 2 provided                                                                                                                  
/usr/include/zmq.hpp:142:16: note: candidate: ‘zmq::message_t::message_t()’                                                                                                                    
  142 |         inline message_t ()                                                                                                                                                            
      |                ^~~~~~~~~                                                                                                                                                               
/usr/include/zmq.hpp:142:16: note:   candidate expects 0 arguments, 2 provided
```

I'm guessing I have the wrong libzmq version? But 4.3.2 seems to be the latest release. And neither the README nor the install docs give much helpful information on dependencies. Or am I supposed to have the header-only zmq bind now? If yes, then why am I not getting cmake errors or missing includes?

vmatare, 2020-05-15:
Taking a closer look at the cmake, I see that various things are not found:
```
Protobuf_DIR:PATH=Protobuf_DIR-NOTFOUND
ZeroMQ_DIR:PATH=ZeroMQ_DIR-NOTFOUND
```
Other things are `NOTFOUND`, too, but these things look like I should be getting cmake errors about it, which I'm not.

caguero, 2020-05-21:
I cannot reproduce this issue. Could you check if in `/usr/include/zmq.hpp` you have a `message_t` constructor with this signature?

caguero, 2020-06-06:
I'm closing this issue, please reopen if the issue persists.

closed 2020-06-06 by caguero