gazebosim/gz-transport #140: [tutorials] link to bitbucket

mohamedsayed18, 2020-05-29:
In the message [tutuorial](https://github.com/ignitionrobotics/ign-transport/blob/master/tutorials/04_messages.md#publisher) the link to download publisher.cc is not working(link to bitbucket)
I think it should be replaced with this link
https://github.com/ignitionrobotics/ign-transport/blob/master/example/publisher.cc

chapulina, 2020-05-29:
Oh I see it's pointing to an inexistent `ign-transport9` branch. I guess we should use `master` until the release.

Do you think you could open a PR updating the broken links?

mohamedsayed18, 2020-05-29:
I opend a pull request and linked it to this issue

chapulina, 2020-05-29:
Thanks! One thing to consider is using `/raw/` instead of `/blob/` like the existing links. Raw should be used if we tell the user to download the files directly, for example with `wget https://github.com...`. But if we just want the user to look at the file, `/blob/` should be ok.

caguero, 2020-05-30:
Thanks, not that you started, do you mind to do the same thing in the other tutorials? I think there are a few extra references to `ign-transport9` branch.

PR #141 (2020-05-30): download link
fix #140
closed 2020-05-30 by caguero