gazebosim/gz-transport #143: [tutorials] Messages, cmake

mohamedsayed18, 2020-06-09:
Error when building the publisher subscriber example, using this [command](https://github.com/ignitionrobotics/ign-transport/blob/master/tutorials/04_messages.md#building-the-code)
`cmake ..`
output
```
CMake Error at CMakeLists.txt:6 (find_package):
  By not providing "Findignition-transport9.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "ignition-transport9", but CMake did not find one.

  Could not find a package configuration file provided by
  "ignition-transport9" with any of the following names:

    ignition-transport9Config.cmake
    ignition-transport9-config.cmake

  Add the installation prefix of "ignition-transport9" to CMAKE_PREFIX_PATH
  or set "ignition-transport9_DIR" to a directory containing one of the above
  files.  If "ignition-transport9" provides a separate development package or
  SDK, be sure it has been installed.
```

caguero, 2020-06-09:
Hi @mohamedsayed18 , how did you install `Ignition Transport`?

mohamedsayed18, 2020-06-09:
sorry this was my mistake, I didn't build it from the source

closed 2020-06-09 by mohamedsayed18