gazebosim/gz-transport #144: [tutorials] installation

mohamedsayed18, 2020-06-09:
I followed the instruction in the [installation page](https://ignitionrobotics.org/api/transport/9.0/installation.html)
`sudo apt-get install libignition-transport9-dev`
Output
`E: Unable to locate package libignition-transport9-dev`
Also I'd like to ask where to find the tutorials web pages in which repo. for example the [messages](https://ignitionrobotics.org/api/transport/9.0/messages.html) tutorial. is it linked to ign-transport/tutorials/04_messages.md?

caguero, 2020-06-09:
Hi @mohamedsayed18 , which Ubuntu version are you using?

All the tutorials can be found here:
https://github.com/ignitionrobotics/ign-transport/tree/master/tutorials

mohamedsayed18, 2020-06-09:
I'm using Ubuntu 20.04 LTS focal 
In the 04_messages.md [publisher.cc](https://raw.githubusercontent.com/ignitionrobotics/ign-transport/master/example/publisher.cc) link is working 
but on the website [publisher.cc](https://bitbucket.org/ignitionrobotics/ign-transport/raw/ign-transport9/example/publisher.cc) is linked to bitbucket

caguero, 2020-06-09:
The version in Ubuntu 20.04 is `libignition-transport4-dev`. See

https://packages.ubuntu.com/search?keywords=ignition-transport&searchon=names&suite=focal&section=all

You're right about the BitBucket link in the tutorials. We'll update it soon.

mohamedsayed18, 2020-06-09:
thank you very much

caguero, 2020-06-09:
Thanks to you for catching the issue!

caguero, 2020-06-09:
It should be fixed now. Refresh your page in case you still see the old BitBucket link.

mohamedsayed18, 2020-06-09:
I checked the publisher.cc and subscriber.cc they still linked to bitbucket

caguero, 2020-06-09:
I'll blame cloudfront and caches if you still don't see it updated. It'll happen eventually. See my screenshot attached with the link to GitHub at the bottom
[image]
.

mohamedsayed18, 2020-06-09:
still the same 
[image]