gazebosim/gz-transport #157: Unable to build on Ubuntu 20.04

pohzhiee, 2020-07-30:
# Description
The following error messages will be shown during build
```
/home/pohzhiee/ign-workspace2/src/ign-transport/src/NodeShared.cc:1072:32: error: ‘class zmq::socket_t’ has no member named ‘set’
 1072 |     this->dataPtr->subscriber->set(zmq::sockopt::subscribe, topic);
      |                                ^~~
/home/pohzhiee/ign-workspace2/src/ign-transport/src/NodeShared.cc:1072:41: error: ‘zmq::sockopt’ has not been declared
 1072 |     this->dataPtr->subscriber->set(zmq::sockopt::subscribe, topic);
      
```

# Cause
This is due to the change in this [commit](https://github.com/ignitionrobotics/ign-transport/commit/d4420b83a01d97e5ef1d5326712eb1c1b44de289)

Note that this is a bug with ZMQ's ubuntu package, and not ign-transport, but it still affects ign-transport nonetheless.

This is due to a bug with the ZMQ versioning in the libzmq3-dev package in ubuntu. Upon installation, the  /usr/include/zmq.hpp will have the following lines:
```c
#define CPPZMQ_VERSION_MAJOR 4
#define CPPZMQ_VERSION_MINOR 7
#define CPPZMQ_VERSION_PATCH 0
```
However, the libzmq shipped with ubuntu 20.04 is 4.3.2 which is reflected in the official ubuntu packages [here](https://packages.ubuntu.com/focal/libzmq3-dev).

# Verification
Run a ubuntu:focal docker container and check that CPPZMQ_VERSION is indeed set to 4.7.0 after installing libzmq3-dev
```sh
docker run -it ubuntu:focal
```
While inside the container
```
apt update && apt install -y libzmq3-dev
grep -i version /usr/include/zmq.hpp
```

# Other notes
The ubuntu bionic libzmq3-dev sets its version correctly so no issues when building with 18.04. This issue only affects 20.04.

# Temporary fixes
Modify /usr/include/zmq.hpp to reflect the correct version

mjcarroll, 2020-08-04:
I can confirm that the `/usr/include/zmq.hpp` file in `focal` has the version set as you describe, which does seem consistent with what the upstream source has. 
 https://github.com/zeromq/cppzmq/blob/89f4d1b187390aeaecc7c0ce0665350dca259753/zmq.hpp#L131-L134
The upstream development being set to 4.7.0 makes sense, as the the last tagged release is 4.6.0

Additionally, looking at the `focal` packaging job, it appears that they are taking the `cppzmq` header at the latest version, which is 4.7.0: https://launchpad.net/ubuntu/+archive/primary/+sourcefiles/zeromq3/4.3.2-2ubuntu1/zeromq3_4.3.2-2ubuntu1.debian.tar.xz

All that to say, I believe that the version number is correct, so we will need to compensate for that.

chapulina, 2020-08-05:
This showed up when adding Focal CI for `ign-transport8` on https://github.com/ignitionrobotics/ign-transport/pull/155.

mjcarroll, 2020-09-01:
Close by #159

closed 2020-09-01 by mjcarroll