gazebosim/gz-transport #192: Conformity of current multicast IP

mbmende, 2020-10-26:
The choice of the multicast IP in https://github.com/ignitionrobotics/ign-transport/blob/1b3a3e0dae80c7b42805d6f54a5a5dfe11f0c27b/include/ignition/transport/Discovery.hh#L1358
should be reconsidered, since it's nonconforming. According to [IANA], "_the range of addresses between 224.0.0.0 and 224.0.0.255, inclusive, is reserved for the use of routing protocols and other low-level topology discovery or maintenance protocols, such as gateway discovery and group membership reporting._" The importance of this range is also the reason, why [RFC4541] requires that "_packets with a destination IP (DIP) address in the 224.0.0.X range which are not IGMP must be forwarded on all ports_" because e.g. routers with missing IGMP join messages could lead to switches splitting up the network. All currently available switches adhere to that recommendation.

The problem with the current choice is that all messages are flooded to the whole network, especially if several instances of gazebo are running with high update rates and >1k packets/s. In our case, with a /22 network and many clients, this leads to a large increase of non directed additional traffic, also to IoT devices and other hosts that don't cope well with massive broadcasts. As this issue might affect a lot of users, even unknowingly, it should be dealt with on a general level instead of in cases where the problem arises.

All other IP ranges defined for multicast are really "multicasted" by switches and hence suited, but since [RFC2365] recommends the 239.255.x.x range for this use, I would suggest to change the hardcoded IP to something like 239.255.0.7 (ROS2 already uses 239.255.0.1 afaik) or make it configurable with a reasonable default value.

[IANA]: https://www.iana.org/assignments/multicast-addresses/multicast-addresses.xhtml#multicast-addresses-13
[RFC4541]: https://tools.ietf.org/html/rfc4541#section-2.1.2
[RFC2365]: https://tools.ietf.org/html/rfc2365#section-6.1

caguero, 2020-11-04:
Thanks for the detailed explanation. It sounds reasonable to make it configurable with a default option in the range `239.255.x.x`.

caguero, 2020-11-12:
@mbmende , see pull request #200 . Could you give it a try?

mbmende, 2020-11-24:
Looks good so far with the latest changes in https://github.com/ignitionrobotics/ign-transport/commit/34f8476bdf1b440a0dccc39bc79d1a5cb6e7be66.

peci1, 2020-12-22:
Good catch... So this is why our site admin was angry with a few gazebos flooding the whole local network :) When I launch Gazebo with IGN_IP=127.0.0.1, then the flooding doesn't occur...

caguero, 2020-12-30:
See pull request #200 .

closed 2020-12-30 by caguero