gazebosim/gz-transport #203: Can't use auto when defining lambda function callbacks for methods like Node::Advertise

adlarkin, 2020-11-19:
As the title states, when trying to call a method like [Advertise](https://ignitionrobotics.org/api/transport/9.0/classignition_1_1transport_1_1Node.html#a04d6c8e855cfdfe972e6399cc4e3a767) with a lambda function callback to create a service, the lambda function must be created before the `Advertise` method, and must also be fully defined - if the lambda function is created using type `auto`, then compilation fails when passing this lambda function into the `Advertise` method. An example can be found [here](https://github.com/ignitionrobotics/ign-gazebo/blob/a5abacfa74db6e8db0218bd32c55980e15cf249b/src/gui/Gui_TEST.cc#L53-L60).

adlarkin, 2021-04-20:
I believe that this is the reason why we must fully define the lambda function before calling `Advertise`: https://stackoverflow.com/a/13359347

Taking a look at [`EntityComponentManager::Each` in `ign-gazebo`](https://github.com/ignitionrobotics/ign-gazebo/blob/ignition-gazebo5_5.0.0/include/ignition/gazebo/EntityComponentManager.hh#L363-L378), I notice that an `identity` is used there, which might be why this issue of pre-defining a lambda function doesn't exist when calling `EntityComponentManager::Each` (I don't see an `identity` being used for [`Node::Advertise`](https://github.com/ignitionrobotics/ign-transport/blob/ignition-transport10_10.0.0/include/ignition/transport/Node.hh#L410-L427)).

arjo129, 2025-09-19:
We should do this for service requests as well.

azeey, 2025-10-16:
https://github.com/gazebosim/gz-transport/pull/382 fixed it for Subscribe. We could use the same approach here.

Ryujiyasu, 2026-02-28:
I'd like to work on this. I'll apply the same approach used in PR #382 (for `Subscribe`) to `Advertise` as well.

closed 2026-03-11 by caguero