gazebosim/gz-transport #220: ign-transport9 debbuilds failing on debian buster

scpeters, 2021-02-10:
## Environment
* OS Version: debian Buster
* debbuild for 9.1.0

## Description
* Expected behavior: expect debbuild to succeed
* Actual behavior: it fails

## Steps to reproduce

1. rebuild the following job: https://build.osrfoundation.org/job/ign-transport9-debbuilder/175/

## Output

~~~
/var/lib/jenkins/workspace/ign-transport9-debbuilder/build/ignition-transport-9.1.0/src/NodeShared.cc:98:16: error: ‘send_flags’ in namespace ‘zmq’ does not name a type
     const zmq::send_flags &_type)
                ^~~~~~~~~~
~~~

Per https://github.com/ignition-tooling/release-tools/issues/297, this may require a port of https://github.com/ignitionrobotics/ign-transport/pull/159

chapulina, 2021-02-10:
I believe #159 is already in `ign-transport9`, I see lots of `IGN_CPPZMQ_POST_4_7_0`s in that branch.

scpeters, 2021-02-10:
> I believe #159 is already in `ign-transport9`, I see lots of `IGN_CPPZMQ_POST_4_7_0`s in that branch.

you're right; I see them too. I don't know what else is needed then