gazebosim/gz-transport #244: Large messages are cropped to 67 MB

peci1, 2021-05-12:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? both 9.1.0

## Description
* Expected behavior: Messages of any reasonable size are transported.
* Actual behavior: E.g. pointclouds from depth cameras with large resolution are not transported whole.

## Steps to reproduce
1. Place an `rgbd_camera` sensor with resolution of 2048x1536 in the scene.
2. Launch ros_ign bridge for the `points` topic of the sensor.
3. The pointcloud that arrives to the ROS subscriber has only 67 MB instead of the expected 75.5 MB.
4. Rviz refuses to render the pointcloud with: `Data size (67108725 bytes) does not match width (2048) times height (1536) times point_step (24). Dropping message.`

## Output

The console of the ros_ign bridge shows these error messages:

```
[libprotobuf ERROR google/protobuf/io/coded_stream.cc:207] A protocol message was rejected because it was too big (more than 67108864 bytes).  To increase the limit (or to disable these warnings), see CodedInputStream::SetTotalBytesLimit() in google/protobuf/io/coded_stream.h.
SubscriptionHandler::CreateMsg() error: ParseFromString failed
```

I understand that having a limit on the message size is a good thing, but it should probably be higher. ROS throws away messages that would be 1 GB and larger according to their metadata.

Or maybe the limit could be configurable, and ros_ign bridge could check if it has to raise it or not.

peci1, 2021-05-12:
Example sensor SDF:

```xml
        <sensor name="camera_marv_fisheye" type="rgbd">
            <topic>/validate/camera_marv_fisheye</topic>
            <update_rate>20</update_rate>
            <camera>
              <horizontal_fov>2.40855</horizontal_fov>
              <image>
                <width>2048</width>
                <height>1536</height>
                <format>R8G8B8</format>
              </image>
              <clip>
                <near>0.025</near>
                <far>300</far>
              </clip>
              <depth_camera>
                <clip>
                  <near>0.01</near>
                  <far>50</far>
                </clip>
              </depth_camera>
            </camera>
          </sensor>
```

Launch ROS-Ign bridge with

    rosrun ros_ign_bridge parameter_bridge /validate/camera_marv_fisheye/points@sensor_msgs/PointCloud2[ignition.msgs.PointCloudPacked"

mjcarroll, 2021-05-17:
> I understand that having a limit on the message size is a good thing, but it should probably be higher.

It appears from the [Protobuf API](https://developers.google.com/protocol-buffers/docs/reference/cpp/google.protobuf.io.coded_stream#CodedInputStream.SetTotalBytesLimit.details) that this should be coded to `INT_MAX (~2GB)` by default.  I couldn't find any evidence we were manipulating that in `ign-transport` or `ros_ign_bridge`.  Did you find somewhere that it was set to 67MB?

Additionally, have you considered using https://github.com/ignitionrobotics/ros_ign/tree/melodic/ros_ign_point_cloud

peci1, 2021-05-17:
No, I haven't found the place where this limit is set. 

ros_ign_point_cloud is not really useful until the ROS-Ign integration in plugins like `gazebo_ros_api_plugin` is resolved. And even then, applications like SubT simulator need to send data to multiple ROS masters, so that would not help either.