gazebosim/gz-transport #263: Topic list not listing subscribed topics

jennuine, 2021-09-24:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
Source both in Dome and Fortress branches

## Description
* Expected behavior: `ign topic -l` to list topics that are being subscribed to
* Actual behavior: Does not list topic until it's published then disappears

## Steps to reproduce

1. `ign gazebo velocity_control.sdf` (this also happens with `diff_drive.sdf` as well)
2. Add topic viewer in gui
3. `ign topic -e -t /model/vehicle_blue/cmd_vel`
4. `ign topic -t "/model/vehicle_blue/cmd_vel" -m ignition.msgs.Twist -p "linear: {x: 0.5}, angular: {z: 0.05}"`

## Output

I came across this when I wanted to check the green vehicle's initial `cmd_vel` and wasn't able to.

[image]

caguero, 2021-09-24:
This is a known limitation. It hasn't been implemented yet.

chapulina, 2021-09-24:
Closed #39 as a duplicate. Usually we'd close the newer issue, but I think that one may be a bit too old.

azeey, 2023-08-29:
Fixed by https://github.com/gazebosim/gz-transport/pull/379

PR #379 (2023-02-08): List subscribed topics when running topic list
# 🎉 New feature

Requires https://github.com/gazebosim/gz-msgs/pull/322

Closes #263 

## Summary

This patch includes the subscribed topics in the output of `gz topic -l`.

## Test it

In one terminal run:

```
gz topic -e -t /foo
```

In another terminal run:

```
gz topic -l
```

Your output should be:
```
/foo
```

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the li [... cut]
closed 2023-08-29 by azeey