gazebosim/gz-transport #286: Prevent log spew when username/password doesn't exist in /etc/passwd

AlexGraberTilton, 2022-01-18:
## Desired behavior
Please stop the spew of "Error getting username: no matching password record.\n"

In commit 09723c46c4e3cf778c2fc36583c941173ecfba5c, there were changes to username retrieval

In the case of no username in the /etc/passwd, the username is set to "error-" + uuid.ToString()
However the error messages introduced in the change cause log spew when no username exists.
The spew causes the console log to be essentially useless, despite the fact that the system still works fine.

One major use case that this impacts is running with the OSRF gazebo docker containers using --user uid:gid
since /etc/passwd does not have the user in such a scenario, the error message repeats endlessly.

## Alternatives considered
1. add a method to prevent multiple error messages for the same uid
2. remove the message entirely

## Implementation suggestion
remove the error message when getpwuid_r passes but the rule is nullptr
backport to 7+

j-rivero, 2022-01-31:
Reading the issue I think it makes sense not to print the error messages since the are valid use cases that triggered them.

@AlexGraberTilton would you mind sending a PR against `ign-transport8` branch? We will review it, merge and merge forward to new versions.

Thanks!

j-rivero, 2022-02-07:
> @AlexGraberTilton would you mind sending a PR against `ign-transport8` branch? We will review it, merge and merge forward to new versions.

Thanks so much for the PR, @AlexGraberTilton . The change should be propagated to newer branches when we forward port them, usually every some weeks.

mjcarroll, 2022-02-08:
Closed via #291

osrf-triage, 2022-03-25:
This pull request has been mentioned on **Gazebo Community**. There might be relevant details there:

https://community.gazebosim.org/t/new-ignition-releases-2022-03-25-fortress-edifice-citadel/1343/1