gazebosim/gz-transport #323: Failed to build and run tutorial `publisher.cc`

marc-wittwer, 2022-06-22:
I followed the tutorial and ran into the following issues while making the script.

https://github.com/gazebosim/gz-transport/blob/ign-transport11/tutorials/04_messages.md

1. I installed the ign transport library from the binary as described [here](https://github.com/gazebosim/gz-transport/blob/ign-transport11/tutorials/02_installation.md)
```
sudo sh -c 'echo "deb http://packages.osrfoundation.org/gazebo/ubuntu-stable `lsb_release -cs` main" > /etc/apt/sources.list.d/gazebo-stable.list'
wget https://packages.osrfoundation.org/gazebo.key -O - | sudo apt-key add -
sudo apt-get update
sudo apt-get install libignition-transport11-dev
```  
2. Create the `ign_transport_tutorial` folder
3. Add the publisher.cc file from the `/examples/publisher.cc` from the `ign-transport11` branch
4. Add the [CMakeLists.txt](https://github.com/gazebosim/gz-transport/blob/ign-transport11/example/CMakeLists.txt) file
5. Run cmake and make: 
```
mkdir build
cd build
cmake ..
make publisher   # it crashes here
```

The `make publisher` command fails with the following error message:
```

[ 50%] Building CXX object CMakeFiles/publisher.dir/publisher.cc.o
In file included from /usr/include/ignition/msgs8/ignition/msgs/detail/PointCloudPackedUtils.hh:24,
                 from /usr/include/ignition/msgs8/ignition/msgs.hh:23,
                 from /home/mwittwer/ign_transport_tutorial/publisher.cc:7:
/usr/include/ignition/msgs8/ignition/msgs/pointcloud_packed.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/pointcloud_packed.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers.  Please
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/pointcloud_packed.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
In file included from /usr/include/ignition/msgs8/ignition/msgs/pointcloud_packed.pb.h:34,
                 from /usr/include/ignition/msgs8/ignition/msgs/detail/PointCloudPackedUtils.hh:24,
                 from /usr/include/ignition/msgs8/ignition/msgs.hh:23,
                 from /home/mwittwer/ign_transport_tutorial/publisher.cc:7:
/usr/include/ignition/msgs8/ignition/msgs/header.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/header.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers.  Please
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/header.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
      |  ^~~~~
In file included from /usr/include/ignition/msgs8/ignition/msgs/header.pb.h:33,
                 from /usr/include/ignition/msgs8/ignition/msgs/pointcloud_packed.pb.h:34,
                 from /usr/include/ignition/msgs8/ignition/msgs/detail/PointCloudPackedUtils.hh:24,
                 from /usr/include/ignition/msgs8/ignition/msgs.hh:23,
                 from /home/mwittwer/ign_transport_tutorial/publisher.cc:7:
/usr/include/ignition/msgs8/ignition/msgs/time.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
   17 | #error This file was generated by an older version of protoc which is
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/time.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
   18 | #error incompatible with your Protocol Buffer headers.  Please
      |  ^~~~~
/usr/include/ignition/msgs8/ignition/msgs/time.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
   19 | #error regenerate this file with a newer version of protoc.
...
...
```

My system: Ubuntu 20.04 in a VM

Thanks for any hints in advance.

mjcarroll, 2022-06-24:
If you have changed the version of protobuf, then you will need to rebuild `msgs` and `transport` (along with all packages above) in order for it to work.  `msgs` contains compiled protobuf-generated code, so if anything changes in the generation/compilation, a rebuild is necessary.

marc-wittwer, 2022-06-30:
@mjcarroll Thanks for your reply.
The msgs were compiled with protobuf v3.6.1 (which corresponds to 3006001). https://github.com/protocolbuffers/protobuf/releases/tag/v3.6.1

Downgrading to this protobuf version solved the error.

I can finally call services such as setting the pose in gazebo :tada: 

Here is the script that I used:
```
#include <iostream>
#include <ignition/msgs.hh>
#include <ignition/transport.hh>

//////////////////////////////////////////////////
int main(int argc, char **argv)
{
  // Create a transport node.
  ignition::transport::Node node;

  // Prepare the input parameters.
  ignition::msgs::Pose req;

  ignition::msgs::Vector3d vec3;
  vec3.set_x(0.0);
  vec3.set_y(0.0);
  vec3.set_z(5.0);

  req.set_name("throwing_object");
  req.set_allocated_position(&vec3);

  ignition::msgs::Boolean rep;
  bool result;
  unsigned int timeout = 5000;

while (true)
  {
      // Request the "/echo" service.
      bool executed = node.Request("/world/my_world/set_pose", req, timeout, rep, result);

  if (executed)
  {
    if (result)
      std::cout << "Response: [" << rep.data() << "]" << std::endl;
    else
      std::cout << "Service call failed" << std::endl;
  }
  else
    std::cerr << "Service call timed out" << std::endl;

    std::this_thread::sleep_for(std::chrono::milliseconds(100));
  }

  
}
```

closed 2022-06-30 by marc-wittwer