gazebosim/gz-transport #371: Installation failure on Ubuntu 20.04

daisukes, 2022-11-20:
I have been encountering this issue since last night on the nightly build task for my ROS project.
It seems like related to the latest version (8.4.0, https://github.com/gazebosim/gz-transport/commit/de204e7624b9e7ba4ddbb012d4ff536c371b7aa1) updated two days ago. 
Could your please look into it. Thank you!
## Environment
* OS Version: Ubuntu 20.04
* binary: apt install

## Description
* Expected behavior: can install without error
* Actual behavior: cannot install with an error

## Steps to reproduce

- `sudo apt-get install libgazebo11-dev`
or
- `sudo apt-get install libignition-transport8-log-dev`

## Output

```
Preparing to unpack .../93-libignition-transport8-log-dev_8.4.0-2~focal_amd64.deb ...
Unpacking libignition-transport8-log-dev:amd64 (8.4.0-2~focal) ...
dpkg: error processing archive /tmp/apt-dpkg-install-9Arral/93-libignition-transport8-log-dev_8.4.0-2~f
ocal_amd64.deb (--unpack):
 trying to overwrite '/usr/include/ignition/transport8/gz/transport/log.hh', which is also in package l
ibignition-transport8-core-dev:amd64 8.4.0-2~focal
Errors were encountered while processing:
 /tmp/apt-dpkg-install-9Arral/93-libignition-transport8-log-dev_8.4.0-2~focal_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

impala454, 2022-11-20:
We have many nightly ROS builds and are receiving this same error on all of them since the 8.4.0 upgrade.  Existing containers also experience this error when apt upgrading.

progtologist, 2022-11-21:
Introduced from https://github.com/gazebosim/gz-transport/commit/9c192aeb3d580077bc6f271fc7102fd84e2e16cf

azeey, 2022-11-21:
Apologies. We're working on a fix.

azeey, 2022-11-21:
Fix in https://github.com/gazebo-release/gz-transport8-release/pull/11 and has been released.

j-rivero, 2022-11-21:
Should be fixed in `8.4.0-3`.

closed 2022-11-21 by j-rivero