gazebosim/gz-transport #385: Missing rosdep key for gz-transport-12

Ryanf55, 2023-02-20:
## Environment
* OS Version: Ubuntu 22.04 (Jammy)
* Source or binary build?

## Description
* Expected behavior: Using rosdep resolves all keys
* Actual behavior: ROSdep is missing keys for gz-transport-12

## Steps to reproduce

Follow instructions for compiling ros_gz here: https://github.com/gazebosim/ros_gz/tree/ros2#compile-ros_gz

## Output

```
ryan@ryan-m93p:~/Documents/gz_ws$ rosdep install -r --from-paths src -i -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros_gz_bridge: Cannot locate rosdep definition for [gz-transport12]
ros_gz_image: Cannot locate rosdep definition for [gz-transport12]
ros_gz_sim: Cannot locate rosdep definition for [gz-math7]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
## The problem

This package has uploaded debians with a new key name, but has not updated the rosdep index, and now other packages are relying on version 11 that does not exist. 
https://github.com/ros/rosdistro/blob/91349fa4751c7a5b62b1129220148a11c693694f/rosdep/base.yaml#L1869-L1875

This was found in this issue: https://github.com/gazebosim/ros_gz/issues/358

Ryanf55, 2023-02-21:
Hey there, I tried being a good citizen and contributing the change to rosdep, but it doesn't work. Would the package maintainer of `gz-transport` be able to assist me with the above PR?

azeey, 2023-02-22:
Please see the updated [documentation](https://gazebosim.org/docs/garden/ros_installation#using-an-specific-gazebo-version-with-ros-2) about using ROS with a specific version of Gazebo including how to add rosdep keys. Since gz-transport12 is not in upstream Ubuntu or packages.ros.org, we can't currently add it to ros/rosdistro.

closed 2023-02-22 by azeey

Ryanf55, 2023-02-22:
Should we then remove it from the `package.xml`?

furquansh, 2025-04-07:
@azeey 
 Im currently running Gazebo Harmonic on Ubuntu 22.04 (Jammy). Im getting the below error 
``
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
bcr_bot: No definition of [gz-transport12] for OS [ubuntu]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

is there any update on whether I can still successfully run this 

Thankyou for your time

Ryanf55, 2025-04-07:
ArduPilot uses this process:
https://ardupilot.org/dev/docs/ros2-gazebo.html#install-gazebo

Specifically, we use the apt sources from here, which allow have rosdep resolve gazebo harmonic when we use ROS 2 humble.
https://github.com/osrf/osrf-rosdep

Seems like the gazebo documentation could give that some more exposure.