gazebosim/gz-transport #387: Add Python bindings

srmainwaring, 2023-02-24:
## Desired behavior

Add Python bindings to `gz-transport` (and `gz-msgs`).

## Alternatives considered

- Use `ros_gz` bridge and the Python support in ROS / ROS2.

## Implementation suggestion

Upstream the bindings for `gz-msgs` and `gz-transport` from `https://github.com/srmainwaring/gz-python`.

## Additional context

From time to time there are requests for Python bindings to control the simulation. A number of the requirements can be met by using the messaging infrastructure to control and interact with the simulation.

- https://community.gazebosim.org/t/python-bindings-for-gazebo-ignition/360
- https://github.com/gazebosim/gz-sim/issues/1905
- https://github.com/gazebosim/gz-transport/pull/386

azeey, 2023-02-24:
Thanks for the implementation suggestion @srmainwaring. This is something we are hoping to address for Harmonic and having your implementation as a starting point will go a long way. We'd also like to address https://github.com/gazebosim/gz-msgs/issues/113, https://github.com/gazebosim/gz-msgs/issues/36, and https://github.com/gazebosim/gz-sim/issues/494 which might have implications for how we implement this.

azeey, 2023-08-29:
This has been implemented in #411

PR #411 (2023-07-24): Python Bindings for Publisher, Subscriber and Service Request features.

# 🎉 New feature

Closes #387 

## Summary

This PR creates the python bindings for the publisher, subscriber and request features from the repository. In order to avoid creating a new dependency to be able to create python bindings from protobuf messages, it was decided to use the methods that publishes, subscribes and make services request with serialized messages, i.e, use the methods `PublishRaw`, `SubscribeRaw` and `RequestRaw`. The bindings for these methods are wrapped by python methods  in order to give the user an [... cut]
closed 2023-08-29 by azeey