gazebosim/gz-transport #400: `gz service -l` does not always output the full list of services

jrutgeer, 2023-05-26:
## Environment
* OS Version:  Ubuntu 22.04
* Gazebo garden, built from source

## Description
* Expected behavior: output of `gz service -l` should always list all available services
* Actual behavior: output of `gz service -l` randomly shows either the full list, or only a subset of the avalable services

## Output

E.g. two consecutive calls of `gz service -l` yield following output:

```
$ gz service -l
/gui/camera/view_control
/gui/camera/view_control/reference_visual
/gui/camera/view_control/sensitivity
/gui/copy
/gui/follow
/gui/follow/offset
/gui/move_to
/gui/move_to/pose
/gui/paste
/gui/screenshot
/gui/view/collisions
/gui/view/com
/gui/view/frames
/gui/view/inertia
/gui/view/joints
/gui/view/transparent
/gui/view/wireframes
/marker
/marker/list
/marker_array
/world/Scara_Picker/level/set_performer
$ gz service -l
/gazebo/resource_paths/add
/gazebo/resource_paths/get
/gazebo/resource_paths/resolve
/gazebo/worlds
/gui/camera/view_control
/gui/camera/view_control/reference_visual
/gui/camera/view_control/sensitivity
/gui/copy
/gui/follow
/gui/follow/offset
/gui/move_to
/gui/move_to/pose
/gui/paste
/gui/screenshot
/gui/view/collisions
/gui/view/com
/gui/view/frames
/gui/view/inertia
/gui/view/joints
/gui/view/transparent
/gui/view/wireframes
/marker
/marker/list
/marker_array
/server_control
/world/Scara_Picker/control
/world/Scara_Picker/control/state
/world/Scara_Picker/create
/world/Scara_Picker/create_multiple
/world/Scara_Picker/declare_parameter
/world/Scara_Picker/disable_collision
/world/Scara_Picker/enable_collision
/world/Scara_Picker/entity/system/add
/world/Scara_Picker/generate_world_sdf
/world/Scara_Picker/get_parameter
/world/Scara_Picker/gui/info
/world/Scara_Picker/level/set_performer
/world/Scara_Picker/light_config
/world/Scara_Picker/list_parameters
/world/Scara_Picker/playback/control
/world/Scara_Picker/remove
/world/Scara_Picker/scene/graph
/world/Scara_Picker/scene/info
/world/Scara_Picker/set_parameter
/world/Scara_Picker/set_physics
/world/Scara_Picker/set_pose
/world/Scara_Picker/set_pose_vector
/world/Scara_Picker/set_spherical_coordinates
/world/Scara_Picker/state
/world/Scara_Picker/state_async
/world/Scara_Picker/system/info
/world/Scara_Picker/visual_config
/world/Scara_Picker/wheel_slip
$
```

It is random, e.g. sometimes many consecutive `gz service -l` each return the full list, but then sometimes a reduced set is reported.

At one time even just one service was reported:

```
$ gz service --list
/gui/follow
$
```

caguero, 2023-07-04:
Thanks for reporting the issue. See https://github.com/gazebosim/gz-transport/pull/415.

mjcarroll, 2023-07-18:
Closed by #415

PR #415 (2023-07-13): Fix topic/service list inconsistency

# 🦟 Bug fix

Fixes #400 

## Summary

Discovery periodically sends topic/service heartbeats to keep all the topics/services advertised by a given node alive in the network. These messages are received by other discovery processes that keep the topics/services up to date or remove them in the absence of heartbeat updates after some time.

In order to get the list of topics/services, `gz` needs to initialize the discovery. This means that discovery should wait some time to receive all the topics/services updates to create the topic/service list. The original [... cut]
closed 2023-07-18 by mjcarroll