gazebosim/gz-transport #404: Compatibility with protobuf 22 (4.22) or later

traversaro, 2023-06-04:
## Desired behavior

It would be great if gz-transport could compile against protobuf 22.

## Alternatives considered

Never update past protobuf 21.

## Implementation suggestion

I am trying to get gz-transport to work with protobuf 23 in conda-forge, and I am getting this error:
~~~
[1/176] Building CXX object src/CMakeFiles/gz-transport12.dir/CIface.cc.o
FAILED: src/CMakeFiles/gz-transport12.dir/CIface.cc.o 
/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_build_env/bin/x86_64-conda-linux-gnu-c++ -Dgz_transport12_EXPORTS -I/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/include -I/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/build/include -I/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/build/core/include -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/gz/utils2 -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/gz/msgs9 -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/gz/math7 -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/uuid -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work=/usr/local/src/conda/libgz-transport12-12.2.0 -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh=/usr/local/src/conda-prefix -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic  -std=c++17 -fPIC -fvisibility=default -msse -msse2 -mfpmath=sse -msse3 -mssse3 -msse4.1 -msse4.2 -fPIC -I/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/uuid -MD -MT src/CMakeFiles/gz-transport12.dir/CIface.cc.o -MF src/CMakeFiles/gz-transport12.dir/CIface.cc.o.d -o src/CMakeFiles/gz-transport12.dir/CIface.cc.o -c /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/src/CIface.cc
In file included from /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/include/gz/transport/NodeShared.hh:42,
                 from /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/include/gz/transport/Node.hh:33,
                 from /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/src/CIface.cc:21:
/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/include/gz/transport/RepHandler.hh:28:10: fatal error: google/protobuf/stubs/casts.h: No such file or directory
   28 | #include <google/protobuf/stubs/casts.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[2/176] Building CXX object src/CMakeFiles/gz-transport12.dir/Clock.cc.o
FAILED: src/CMakeFiles/gz-transport12.dir/Clock.cc.o 
/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_build_env/bin/x86_64-conda-linux-gnu-c++ -Dgz_transport12_EXPORTS -I/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/include -I/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/build/include -I/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/build/core/include -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/gz/utils2 -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/gz/msgs9 -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/gz/math7 -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/uuid -fvisibility-inlines-hidden -fmessage-length=0 -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work=/usr/local/src/conda/libgz-transport12-12.2.0 -fdebug-prefix-map=/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh=/usr/local/src/conda-prefix -O3 -DNDEBUG  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Winit-self -Wswitch-default -Wmissing-include-dirs -pedantic  -std=c++17 -fPIC -fvisibility=default -msse -msse2 -mfpmath=sse -msse3 -mssse3 -msse4.1 -msse4.2 -fPIC -I/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeh/include/uuid -MD -MT src/CMakeFiles/gz-transport12.dir/Clock.cc.o -MF src/CMakeFiles/gz-transport12.dir/Clock.cc.o.d -o src/CMakeFiles/gz-transport12.dir/Clock.cc.o -c /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/src/Clock.cc
In file included from /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/include/gz/transport/NodeShared.hh:42,
                 from /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/include/gz/transport/Node.hh:33,
                 from /home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/src/Clock.cc:26:
/home/conda/feedstock_root/build_artifacts/gz-transport12_1685541431637/work/include/gz/transport/RepHandler.hh:28:10: fatal error: google/protobuf/stubs/casts.h: No such file or directory
   28 | #include <google/protobuf/stubs/casts.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~~

This header was used as the `google::protobuf::down_cast` function was used in the code.

[... cut ...] `). As the history is so complicated to navigate, it is a bit difficult for me to understand how to properly replace `google::protobuf::down_cast` when using protobuf 22 or later, so any input on this is welcome. Perhaps just using const_cast or some other cast is sufficient?

## Additional context

Example of existing usages of `google::protobuf::down_cast`:
* https://github.com/gazebosim/gz-transport/blob/c958aec699bdf97690be0ad84a6e8efa6ec7b308/include/gz/transport/SubscriptionHandler.hh#L214

traversaro, 2023-06-04:
Note that this problem may soon hit homebrew: https://github.com/Homebrew/homebrew-core/pull/131405 .

bmaxdk, 2023-08-12:
It seems like the issue is related to a change in the Google Protocol Buffers library, where certain functionalities were removed or replaced. In your particular case, it seems like the `google::protobuf::down_cast` method has been removed and replaced with `::google::protobuf::internal::DownCast`.

try replace this section of `down_cast` method
`auto ptr = google::protobuf::down_cast<Derived*>(basePtr);`

to

`auto ptr = ::google::protobuf::internal::DownCast<Derived*>(basePtr);`

Hope this helped

traversaro, 2023-08-12:
Hello @bmaxdk , this was solved in https://github.com/gazebosim/gz-transport/pull/405, thanks for the comment anyway!

closed 2023-08-12 by traversaro