gazebosim/gz-transport #412: :woman_farmer:  Build regression on `gz-transport-13-debbuilder` error with ` fatal error: gz/math/AxisAlignedBox.hh: No such file or directory`

claraberendsen, 2023-06-23:
## Environment
* OS Version:  linux-drogon.nv.focal-cba5e3fd
* Source or binary build?
[#288](https://build.osrfoundation.org/job/gz-transport13-debbuilder/288/)
  
  

## Description

Seems this error has occurred before since build [#284](https://build.osrfoundation.org/job/gz-transport13-debbuilder/284/) and it seems it has the behavior where it would fail and on the next build pass. 
Could not determine what introduced this, it could possibly be something related to the build of gz-msgs10? 

## Output

```
usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/gz/msgs10/gz/msgs/Utility.hh:24:10: fatal error: gz/math/AxisAlignedBox.hh: No such file or directory
   24 | #include <gz/math/AxisAlignedBox.hh>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
�[33mautopkgtest [11:09:54]: test build: -----------------------]
�[0m�[33mautopkgtest [11:09:54]: test build:  - - - - - - - - - - results - - - - - - - - - -
�[0mbuild                FAIL non-zero exit status 1
�[33mautopkgtest [11:09:54]: test build:  - - - - - - - - - - stderr - - - - - - - - - -
�[0mIn file included from /usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/gz/transport13/gz/transport/Discovery.hh:71,
                 from /usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/gz/transport13/gz/transport.hh:26,
                 from gztest.cc:1:
/usr/lib/x86_64-linux-gnu/pkgconfig/../../..//include/gz/msgs10/gz/msgs/Utility.hh:24:10: fatal error: gz/math/AxisAlignedBox.hh: No such file or directory
   24 | #include <gz/math/AxisAlignedBox.hh>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated
```

j-rivero, 2023-07-07:
Waiting https://github.com/gazebo-release/gz-msgs10-release/pull/3 to land before debugging

claraberendsen, 2023-07-19:
Occurred today again. See https://build.osrfoundation.org/job/gz-transport13-debbuilder/343. It seems to be happening only when it rans on drogon, not on the normal agents.

azeey, 2023-08-03:
Potential fix in https://github.com/gazebosim/gz-cmake/pull/374