gazebosim/gz-transport #446: Add -h option to gz topic command line

TheotimeBalaguer, 2023-09-27:
## Desired behavior
There is no straight-forward way of getting the number of messages per second published on a gazebo transport topic. Basically, this is an extension of the `gz topic` command with a new option (I propose `-h` for consistency with gazebo-classic) that prints the number of received messages on a particular topic. The functioning would be very close to the `ros2 topic hz <topic_name>` command of the ROS middleware.

## Alternatives considered
There could be an alternative using a mix of `watch`, `grep`, `wc -l` but that is not easily usable for everyone.

## Implementation suggestion
- The user runs `gz topic -h -t /my/topic/name` 
- A Gazebo transport node activate and subscribe to the given topic, count the number of receptions, and print each second to the standard output relevant information being the number of received message during this second, the average time between messages, etc. (to be discussed) 

## Additional context
For reference, the output of a ros2 hz command :
[image]

TheotimeBalaguer, 2023-09-28:
For the curious out there, the VERY clumsy one-liner to have an (imprecise !) idea of the number of messages per second : 
`watch -n 1 "( timeout 1s gz topic -e -t /model/px4vision_lidar_2/odometry | grep nsec || true ) | wc -l"`

azeey, 2023-11-30:
`-h` is already used for "help", so we should probably use a different flag.