gazebosim/gz-transport #473: `gz service` should not require `--reqtype` and `--reptype`

azeey, 2024-01-26:
## Desired behavior
`gz service -r` automatically determines the Request and Response message types.  So instead of:
```
gz service -s /gazebo/worlds --reqtype gz.msgs.Empty --reptype gz.msgs.StringMsg_V -r 'unused: true' --timeout 2000
```
I'd like to run:
```
gz service -s /gazebo/worlds  -r 'unused: true' --timeout 2000
```

## Alternatives considered
n/a

## Implementation suggestion
`gz service -i` is able to retrieve the request and response types so it should be possible to use the same mechanism when making service calls via `gz service -r`.

## Additional context
n/a

sam-ovalle, 2025-10-31:
Ill be looking over this one with @haresh-seenivasagan (github.com/haresh-seenivasagan)

ielson, 2026-03-12:
Hey @azeey PR is ready for review:
https://github.com/gazebosim/gz-transport/pull/829