gazebosim/gz-transport #474: `gz service -r` should not require specifying a dummy message if the request type is `Empty`

azeey, 2024-01-26:
## Desired behavior
Currently, `gz service -r` requires specifying `unused: true` on requests with an `Empty` request type. This is not well documented and seems unintuitive.  It would be great if the user is able to provide an empty message (`gz service -r ' ' ...`) or no value at all (`gz service -r ...`). 

Together with #473, it would be nice to be able to run
```
gz service -s /gazebo/worlds -r --timeout 2000
```
instead of
```
gz service -s /gazebo/worlds --reqtype gz.msgs.Empty --reptype gz.msgs.StringMsg_V -r 'unused: true' --timeout 2000
```

## Alternatives considered
n/a

## Implementation suggestion
Handle the case where `reqtype` is `Empty` so that an empty message is considered valid.

## Additional context
n/a

caguero, 2024-03-21:
@azeey , What do you think about allowing `-r ""`? The option `-r` is configured as a parameter that accepts a value. Not sure if we can make the value optional.

mjcarroll, 2024-03-21:
I believe you can add `->expected(0,1)` to make it take 0 or 1 arguments

azeey, 2024-03-21:
I'm okay with `-r ""`, but if @mjcarroll's suggestion works, that would be even better.

caguero, 2024-03-21:
> I believe you can add `->expected(0,1)` to make it take 0 or 1 arguments

Thanks, it works!

closed 2024-05-31 by azeey