gazebosim/gz-transport #476: Provide a tool to diagnose network setup issues

azeey, 2024-01-29:
## Desired behavior
Many new users run into problems using Gazebo due to improper network configuration. It would be great if we can provide a tool to diagnose the problem. As a first step, we can provide a tool like
https://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html#enable-multicast that checks if there is a UDP multicast problem blocking discovery.

## Alternatives considered
n/a

## Implementation suggestion
This could be a `gz` command.

## Additional context
Related issues:
https://github.com/gazebosim/gz-sim/issues/1879
https://github.com/gazebosim/gz-sim/issues/2285

mapi-ng, 2024-08-11:
Hi @azeey, thank you for bringing this issue.

I have just recently installed ROS2 and Gazebo and I ran into this issue.

I think just having a timeout and an error message with suggestion to enable multicast would be of great help to new users. As a user trying to troubleshoot this kind of issue it would not be my first thought to run a separate `gz` command for network diagnosis - here is why.

Before I realized that it was sufficient to [Enable multicast](https://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html#enable-multicast), I had troubles understanding what the real problem was. My first thought was about graphics driver configuration.

Only after using verbose output with `-v 4` debug level did I understand that the issue was rather network related and only then was I able to find this related issue: https://github.com/gazebosim/gz-sim/issues/2285

Ujjawal-prasad, 2025-01-08:
Thank you soo much , it worked

ghak, 2025-02-12:
For those who still have trouble even with multicast enabled, the issue might be related to your firewall. If you’re running the simulation on a single computer, you don’t necessarily have to disable your firewall entirely. Instead, try binding the Gazebo IP to the loopback interface by setting:
export GZ_IP=127.0.0.1

Padmaprabhan2004, 2025-10-30:
Thanks! It finally worked