gazebosim/gz-transport #509: :farmer: Gz-transport12-homebrew `UNIT_gz_TEST gzTest.TopicInfo` failing

Crola1702, 2024-07-12:
## Environment
* OS Version: MacOS
* Source gz-transport12 build

## Steps to reproduce
1. Run a build in [gz-transport-12](https://build.osrfoundation.org/job/gz_transport-ci-gz-transport12-homebrew-amd64)
2. See UNIT_gz_TEST fail

## Output
Reference build: https://build.osrfoundation.org/job/gz_transport-ci-gz-transport12-homebrew-amd64/59/

Test regression:
* [UNIT_gz_TEST.xml.[failed-to-read]](https://build.osrfoundation.org/job/gz_transport-ci-gz-transport12-homebrew-amd64/59/testReport/UNIT_gz_TEST/xml/_failed_to_read_/)

Log output:
```
33: [ RUN      ] gzTest.TopicInfo
33: /Users/jenkins/jenkins-agent/workspace/gz_transport-ci-gz-transport12-homebrew-amd64/gz-transport/src/cmd/gz_TEST.cc:124: Failure
33: Value of: output
33:   Actual: false
33: Expected: true
33: OUTPUT[�\0`\0\0Д\0`\0\0z���\0\0�\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0�Ս��\0\0\0p��\0\0@��\0\0\0\0\0\0\0\0\0\0�@���\0\0�] Size[80]. Expected Size=50
33: 
33: [  FAILED  ] gzTest.TopicInfo (26576 ms)
```

And then:

```
Failed to read test report file /Users/jenkins/jenkins-agent/workspace/gz_transport-ci-gz-transport12-homebrew-amd64/build/test_results/UNIT_gz_TEST.xml
org.dom4j.DocumentException: Invalid byte 1 of 1-byte UTF-8 sequence.
```

Repository doesn't have any new changes.

Crola1702, 2024-09-16:
This is still a consistent issue in gz-transport-11

azeey, 2024-11-05:
The ign-transport11 version of this bug was fixed in #551. But this is for Garden, which is going EOL this month. It is not trivial to forward port #551 and since this is really a timing issue in the test and not a bug in the gz-transport library, I'm inclined to close this issue without fixing it. If we have the same issue appear in gz-transport13, we can fix it there.

closed 2024-11-11 by azeey