gazebosim/gz-transport #510: :farmer: Gz-transport11-homebrew `UNIT_gz_TEST ignTest.TopicInfo` failed

Crola1702, 2024-07-15:
## Environment
* OS Version: MacOS
* Source build

## Steps to reproduce
1. Run a build in [gz-transport11](https://build.osrfoundation.org/job/gz_transport-ci-ign-transport11-homebrew-amd64/)
2. See `ignTest.TopicInfo` fail

## Output

Reference build: https://build.osrfoundation.org/job/gz_transport-ci-ign-transport11-homebrew-amd64/59/consoleFull

Log output:
```
33: [ RUN      ] ignTest.TopicInfo
33: /Users/jenkins/jenkins-agent/workspace/gz_transport-ci-ign-transport11-homebrew-amd64/gz-transport/src/cmd/gz_TEST.cc:135: Failure
33: Value of: infoFound
33:   Actual: false
33: Expected: true
33: OUTPUT[No publishers on topic [/foo]
33: ] Size[30]. Expected Size=50
33: 
33: /Users/jenkins/jenkins-agent/workspace/gz_transport-ci-ign-transport11-homebrew-amd64/gz-transport/src/cmd/gz_TEST.cc:138: Failure
33: Value of: output.find("ignition.msgs.Vector3d") != std::string::npos
33:   Actual: false
33: Expected: true
33: [  FAILED  ] ignTest.TopicInfo (26598 ms)
```

Might be related to #509

scpeters, 2024-07-15:
I suspect that this is a flaky test. It is now failing consistently on macOS since [build 58](https://build.osrfoundation.org/job/gz_transport-ci-ign-transport11-homebrew-amd64/58/), though it still passes on Ubuntu. There were no changes in the `ign-transport11` branch from [build 57](https://build.osrfoundation.org/job/gz_transport-ci-ign-transport11-homebrew-amd64/57/) to [build 58](https://build.osrfoundation.org/job/gz_transport-ci-ign-transport11-homebrew-amd64/58/), and I compared the log files and did not see any significant changes in dependencies. Looking at the [code for this failing test case](https://github.com/gazebosim/gz-transport/blob/b0328a184ca52227b70286f437333a0afc0baf02/src/cmd/gz_TEST.cc#L109-L142), I see it starts a publisher and waits a few seconds, which seems like a race condition to me.

azeey, 2024-11-05:
Fixed by #551.

PR #551 (2024-11-05): Fix flakiness in the TopicInfo test
# 🦟 Bug fix

Fixes #510 

## Summary
The flakiness appears to be a timing issue. The fix here involves running the loop that waits for the input messages in a separate thread before starting the publisher.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test co [... cut]
closed 2024-11-05 by azeey