gazebosim/gz-transport #511: ModuleNotFoundError: No module named 'gz'

Andrii0607, 2024-07-18:
I am trying to execute the "publisher.py" script, which is in the "python" directory. I have installed "libgz-transport10-dev" and "libgz-transport13-dev" using apt. But whenever I execute the script 
$ python3 publisher.py
I get 
ModuleNotFoundError: No module named 'gz'
error.
Do I have to install some additional packages?
I am using Ubuntu24.

azeey, 2024-07-18:
Yeah, you'll need to install `python3-gz-transport13`. 

It would be great to include some installation instructions in https://gazebosim.org/api/transport/13/python.html. If you can help out with that, i.e, update https://github.com/gazebosim/gz-transport/blob/gz-transport13/tutorials/06_python_support.md, I would greatly appreciate it.

Andrii0607, 2024-07-19:
@azeey 
Thank you! Your answer helped me a lot!
OK. I will update the document and open a pull request.

Andrii0607, 2024-07-21:
@azeey 
It seems I don’t have a permission to create a new branch. Can you please grant me the permission?

mjcarroll, 2024-07-22:
>It seems I don’t have a permission to create a new branch. Can you please grant me the permission?

You can fork the repo into your own namespace and make changes there before opening a pull request.

PR #516 (2024-08-13): Added Prerequisites section in file
…rt13

# 🦟 Bug fix

Fixes #511 

## Summary
Added Prerequisites section specifying to install the required library

## Checklist
- [x] Signed all commits for DCO
- [x] Updated documentation (as needed)
- [x] While waiting for a review on your PR, please help review [another open pull request](https://github.com/pulls?q=is%3Aopen+is%3Apr+user%3Agazebosim+archived%3Afalse+) to support the maintainers

**Note to maintainers**: Remember to use **Squash-Merge** and edit the commit message to match the pull request summary while retaining `Signed-off-by` messages.
closed 2024-08-13 by azeey