gazebosim/gz-transport #544: Gazebo Server on Docker Container in EC2 instance & Gazebo Client on MacOS

Scott-Hickmann, 2024-09-24:
Hello,

I have Gazebo Harmonic installed on both my Mac and my AWS EC2 instance Docker container. I am trying to run Gazebo on that EC2 instance, but unfortunately it is headless. I first tried setting up X11 forwarding, but this fails because of a wrong OpenGL version on my Mac.

I then tried instead to make the EC2 instance run only the server of Gazebo, and on my Mac run the client only. However, these two programs are running on different networks. I hence tried using VSCode TCP forwarding to forward all ports from the server to my Mac's localhost, but still encountered an issue with discovery.

I ran the following on my Mac Terminal:
```bash
GZ_IP=127.0.0.1 GZ_PARTITION=127.0.0.1:ros GZ_VERBOSE=1 gz sim -v 4 -g
```
and the following on a VSCode Terminal (so that TCP port forwarding happens automatically) on my EC2 instance within the Docker container:
```bash
GZ_IP=127.0.0.1 GZ_PARTITION=127.0.0.1:ros GZ_VERBOSE=1 gz sim shapes.sdf -s -v
```

The problem is that the UDP discovery process doesn't happen properly, since VSCode only forwards TCP, and hence the client keeps printing:
```
[GUI] [Dbg] [Gui.cc:343] GUI requesting list of world names. The server may be busy downloading resources. Please be patient
```
I could not find any straightforward way to get around the UDP discovery. UDP forwarding does not seem to be trivial? Do you know how to fix this? What is the recommended way to have a Gazebo server and client running on two different networks?

Thank you for your help,
Scott

rom341, 2024-11-22:
Hi, 
I am also trying to run the gazebo client and server on aws and a real computer and here is what can help to solve this problem: 

If multicast traffic does not pass between the client and the server, then it is worth using a vpn that handle udp multicast to let server and client talk with each other.

If you dont want to use vpn, try the [GZ_RELAY](https://gazebosim.org/api/transport/12/relay.html) variable on each client pointed to server ip.

Please, let me know if this helps,
Roman

azeey, 2024-12-16:
Thanks for the issue, but this is a type of question that is best asked on https://robotics.stackexchange.com/. I encourage you to post your question there and if you find that there's a specific bug or feature request, come back and create a new issue here. In the mean time, I'll go ahead and close this.

closed 2024-12-16 by azeey

RobinSchmid7, 2025-03-09:
@Scott-Hickmann Did you find a way to solve this?

rom341, 2025-03-09:
I am using OpenVPN network to connect gz server and client. I have successfully connected gz client to remote server. Configuration for thisi described here https://github.com/gazebosim/gz-sim/issues/2698#issuecomment-2614834921