gazebosim/gz-transport #560: cppzmq-dev has not been installed alongside gz-transport

romulogcerqueira, 2024-12-05:
In Ubuntu 24.04 and later, the C++ bindings for ZeroMQ have been moved to the `cppzmq-dev` [package](https://packages.ubuntu.com/search?keywords=cppzmq-dev&searchon=names&suite=all&section=all).

However, this package is not installed as a dependency alongside `gz-transport`, which relies on these bindings. This oversight may lead to build or runtime issues for users who expect `gz-transport` to work out of the box.

azeey, 2024-12-05:
Looks like there was a fix for this in https://github.com/gazebo-release/gz-transport13-release/pull/9, but it wasn't ported to gz-transport14.

azeey, 2025-01-06:
@j-rivero would you mind porting the changes to gz-transport14?