gazebosim/gz-transport #562: error message: "error sending response: Host unreachable" even if the service call behaved as exepected

paoloelle, 2024-12-09:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build? Binary Gazebo Fortress

## Description
* Expected behavior: Some message that tells me if the service call behaved as expected even if an error occurred in sending the response to the service
* Actual behavior:
I'm just implementing a simple scenario where I have a robot and a couple of simple objects that need to be moved in the arena with a certain frequency. I have a bash script that allows me to do that. The problem that I'm facing is that, even if the robot and the objects are moved as expected, I keep seeing the message `[ruby $(which ign) gazebo-1] NodeShared::RecvSrvRequest() error sending response: Host unreachable` when the timeout for the service is triggered (`Service call timed out`). 

## Steps to reproduce

1. Run a simple simulation with some objects
2. Change the position of the objects continuously

## Output
This is the output that I get when the timeout for the service call expires, even if the object is moved to the new position as expected: 

- `[ruby $(which ign) gazebo-1] NodeShared::RecvSrvRequest() error sending response: Host unreachable`
- `Service call timed out`

azeey, 2024-12-20:
@caguero would you be able to take a look?

paoloelle, 2024-12-21:
I’m trying to run the same example with more output for debugging. Could there be a maximum number of `ign service call` commands that can be executed in a single simulation run? I tried the same  also with Gazebo Harmonic and ROS 2 Jazzy but the problem seems exactly the same.

paoloelle, 2024-12-22:
Follow-up of this

> I’m trying to run the same example with more output information for debugging. Could there be a maximum number of `ign service call` commands that can be executed in a single simulation run? I tried the same also with Gazebo Harmonic and ROS 2 Jazzy but the problem seems exactly the same.

These are the results of the experiment mentioned above: it consists of stopping the simulation, moving the TurtleBot4 robot, moving the objects, and letting the simulation run for 10 seconds. The experiment is shown in the following video. 

https://github.com/user-attachments/assets/f8050a79-fb7c-44dd-aae1-51315dcff70b

I executed this simulation for ~24 hours,  and these are the results that I got. In the following plots please don't take into account the 25th hour. 

[image]
[image]
[image]

As can be noticed from the plots there is both a general decrease in the number of services called per hour and the successful rate per hour.

paoloelle, 2025-01-07:
Please refer to #564 

> Follow-up of this
> 
> > I’m trying to run the same example with more output information for debugging. Could there be a maximum number of `ign service call` commands that can be executed in a single simulation run? I tried the same also with Gazebo Harmonic and ROS 2 Jazzy but the problem seems exactly the same.
> 
> These are the results of the experiment mentioned above: it consists of stopping the simulation, moving the TurtleBot4 robot, moving the objects, and letting the simulation run for 10 seconds. The experiment is shown in the following video.
> 
>  simple_sim-2024-12-20_09.56.47.mp4 
> 
> I executed this simulation for ~24 hours, and these are the results that I got. In the following plots please don't take into account the 25th hour.
> 
> ![Total Service Calls Per Hour](https://private-user-images.githubusercontent.com/61319213/397981153-85906e37-c60a-438d-a5d5-c9f04af26820.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYyNDE3MjAsIm5iZiI6MTczNjI0MTQyMCwicGF0aCI6Ii82MTMxOTIxMy8zOTc5ODExNTMtODU5MDZlMzctYzYwYS00MzhkLWE1ZDUtYzlmMDRhZjI2ODIwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA3VDA5MTcwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZjMzIwOTYxYTBiMDY0YjAxY2MwODUxOWQyNTM3OGRkNWYxODk2NzJjMzRiYjJkNjk3NTUxZjUwYTE5MjQzYTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.jHsH4gLghP0JfXA6k5NdgV1b8h-8ESfkk4ct0E_4oO0) ![Failed Service Calls Per Hour](https://private-user-images.githubusercontent.com/61319213/397981165-73ef47cb-855d-476f-b6ce-0d4a139e3688.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYyNDE3MjAsIm5iZiI6MTczNjI0MTQyMCwicGF0aCI6Ii82MTMxOTIxMy8zOTc5ODExNjUtNzNlZjQ3Y2ItODU1ZC00NzZmLWI2Y2UtMGQ0YTEzOWUzNjg4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA3VDA5MTcwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZlMDI1NTUwMDdiMTI4MmRjY2U4MTllOWUzYzI2ZmIyYmYxOGRmMWUxMjY4MDI2ODAzNTM3MTRjOGMyOTdhMmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.TukNjd0rOAlhUufwBgJ1NuDVPwiJkTkF_khfQzhbE2E) ![Success Percentage of Service Calls Per Hour](https://private-user-images.githubusercontent.com/61319213/397981169-f4c2072f-1aeb-4026-906a-b011fccff31f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYyNDE3MjAsIm5iZiI6MTczNjI0MTQyMCwicGF0aCI6Ii82MTMxOTIxMy8zOTc5ODExNjktZjRjMjA3MmYtMWFlYi00MDI2LTkwNmEtYjAxMWZjY2ZmMzFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA3VDA5MTcwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZlNWNkNTc3NjRiN2Q2Mjg1MGIxMTllZGEzNzQ1NGU4N2NhOGM2YmIxZDNkZDMyN2EyMDQzYjBiMDhkYTUyMDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.ESWxq_LYxtK1epjvMYXNEXkRsBMraqZFrjqZ0o9srUg)
> 
> As can be noticed from the plots there is both a general decrease in the number of services called per hour and the successful rate per hour.
> 
> Please let me know if I have to open another issue related to this or we can keep this one

paoloelle, 2025-01-07:
Please refer to #564.

closed 2025-01-07 by paoloelle

paoloelle, 2025-01-07:
Closed as duplicate of #564