gazebosim/gz-transport #564: Increased Failure Rate of Gazebo Service Calls During Long Simulations with ROS 2

paoloelle, 2025-01-01:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
Tested with Gazebo Fortress and with Gazebo Harmonic (using Apptainer containerization)

## Description
I'm running an evolutionary algorithm using ROS 2 Humble and Gazebo Fortress (as I said before I tested it also with ROS 2 Jazzy and Gazebo Harmonic via Apptainer containerization). To do this, I need to call Gazebo services to stop/start the simulation and move various objects in the environment, including my robot. What I noticed is that after calling these commands, the simulations' performance slows down, especially because the service call often fails.

To test this undesired behavior, I implemented a simple simulation scenario, without any evolutionary algorithm on top of it. This simulation consists of just stopping the simulation, moving the robot and the objects, and restarting the simulation for 10 seconds. An example is shown in the video below.

https://github.com/user-attachments/assets/06251eb4-0ea5-4199-9092-c679ef3264ce

I let this simple simulation run for 24 hours and plotted the related results.
These show the total number of service calls for each hour of simulation, the service calls that didn't give back a successful answer, and the percentage of service calls that gave a successful answer. Please ignore the hour 25.

[image]

[image]

[image]

azeey, 2025-01-07:
Thanks for the detailed investigation into this issue. We're taking a look at the problem. However, it looks like this is a duplicate of https://github.com/gazebosim/gz-transport/issues/562. If so, do you mind closing one of them?

caguero, 2025-01-09:
Is there any change you could share the code you use to reproduce this issue? Thanks!

paoloelle, 2025-01-10:
Yes, sure. These folder contains  the files that I used for the environment:
[sdf_files.zip](https://github.com/user-attachments/files/18373173/sdf_files.zip)

This is the bash script that I used to start/stop the simulation and move the robot and the objects:
```bash
#!/bin/bash

START_TIME=$(date +%s)

log() {
    CURRENT_TIME=$(date +%s)
    ELAPSED=$((CURRENT_TIME - START_TIME))
    HOURS=$((ELAPSED / 3600))
    MINUTES=$(((ELAPSED % 3600) / 60))
    SECONDS=$((ELAPSED % 60))
    printf "[$(date +"%H:%M:%S")][%02d:%02d:%02d] %s\n" "$HOURS" "$MINUTES" "$SECONDS" "$1"
}

# SIMPLE SCRIPT TO MOVE CONTINUOSLY OBJECTS AND ROBOT IN SIMULATION

#simulation_id=0

COUNTER=0 # counter for the call to ign services

#export ROS_DOMAIN_ID=$simulation_id && export IGN_PARTITION=$simulation_id

# export IGN_VERBOSE=1 # print more debugging info

#export IGN_IP=127.0.0.1 # FIXME not sure if it fixes the problem of the export Exception sending a multicast message:Network is unreachable
                        # this command should not be necessary

fastdds shm clean # https://github.com/eProsima/Fast-DDS/issues/2003#issuecomment-1160245640
log ""

while true; do

    #ros2 param set /ann_controller genome_id test

    #ros2 param set /ann_controller update_genome_id True
    
    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"
    log "[DEBUG] Start simulation"
    ign service -s /world/arena/control --reqtype ignition.msgs.WorldControl --reptype ignition.msgs.Boolean --timeout 5000 --req 'pause: true'

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER" 
    log "[DEBUG] Respawn turtlebot"
    # ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'turtlebot4' position: { x: 1, y: 1, z: 1 } orientation: { x: 0, y: 0, z: 0, w: 1}" --timeout 2000
    ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req 'name: "turtlebot4" position: { x: 1.0, y: 1.0, z: 0.2 } orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }' --timeout 5000
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER" 
    log "[DEBUG] Respawn object1"
    ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object1' position: { x: -1.5, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER" 
    log "[DEBUG] Respawn object2"
    ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object2' position: { x: -1.2, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000   
    sleep 1
    
    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"    
    log "[DEBUG] Respawn object3"
    ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object3' position: { x: -0.8, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER" 
    log "[DEBUG] Respawn object4"
    ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object4' position: { x: 0.2, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000   
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"   
    log "[DEBUG] Respawn object5"
    ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object5' position: { x: 0.8, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000 
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"   
    log "[DEBUG] Respawn object6"
    ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object6' position: { x: 1.1, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000 
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"
    log "[DEBUG] Respawn object7"
    ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object7' position: { x: 1.3, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"
    log "[DEBUG] Stop simulation"
    ign service -s /world/arena/control --reqtype ignition.msgs.WorldControl --reptype ignition.msgs.Boolean --timeout 5000 --req 'pause: false'

    sleep 10

done
```

I didn't post the code to simulate the [TurtleBot 4](https://turtlebot.github.io/turtlebot4-user-manual/software/turtlebot4_simulator.html) robot since there are various folders to download for the simulation.

paoloelle, 2025-01-14:
@caguero  just a follow-up. I also tried the setup where I move only the position of the objects without launching anything related to the Turtlebot 4, and the problem still arises. I also tried moving only the position of the Turtlebot 4 without moving the objects and, also in this case, the same problem occurs.

caguero, 2025-01-14:
Thanks, that's useful information.

azeey, 2025-01-14:
@paoloelle have you tried it with a simpler robot, say an `rrbot` (https://github.com/gazebosim/ros_gz_project_template/blob/main/ros_gz_example_description/models/rrbot/model.sdf)?

paoloelle, 2025-01-14:
@azeey I didn't try with a simpler robot, anyway as I mentioned here

> @caguero just a follow-up. I also tried the setup where I move only the position of the objects without launching anything related to the Turtlebot 4, and the problem still arises. I also tried [...]

I didn't spawn any robot at all.

paoloelle, 2025-01-16:
@azeey I tried with the `rrobot` as well (I set it as a `static` entity) but the same happend. The output from the terminal where I launch Gazebo is `NodeShared::RecvSrvRequest() error sending response: Host unreachable` when I call a service and I receive back a `Service call timed out`.

paoloelle, 2025-01-24:
@caguero @azeey any update? 
Thanks in advance

paoloelle, 2025-01-27:
@caguero @azeey I also tried another approach by writing [two plugins](https://github.com/paoloelle/world_control_plugin.git) that don't use services) to modify the status of the simulation (paused/running) or the models' position:

- `PauseResum` reads from the topic a boolean value to resume/pause simulation
- `Movemodel` follows the same idea as `PauseResume` but here we have the name of the model and the new desired position. 

I tested these new plugins in the same environment provided [here](https://github.com/gazebosim/gz-transport/issues/564#issuecomment-2582101138) and the code to automatize the pause/resume commands and the movement of the objects' position is listed below
```bash
#!/bin/bash

START_TIME=$(date +%s)

log() {
    CURRENT_TIME=$(date +%s)
    ELAPSED=$((CURRENT_TIME - START_TIME))
    HOURS=$((ELAPSED / 3600))
    MINUTES=$(((ELAPSED % 3600) / 60))
    SECONDS=$((ELAPSED % 60))
    printf "[$(date +"%H:%M:%S")][%02d:%02d:%02d] %s\n" "$HOURS" "$MINUTES" "$SECONDS" "$1"
}

# SIMPLE SCRIPT TO MOVE CONTINUOSLY OBJECTS AND ROBOT IN SIMULATION

#simulation_id=0

COUNTER=0 # counter for the call to ign services

#export ROS_DOMAIN_ID=$simulation_id && export IGN_PARTITION=$simulation_id

# export IGN_VERBOSE=1 # print more debugging info

#export IGN_IP=127.0.0.1 # FIXME not sure if it fixes the problem of the export Exception sending a multicast message:Network is unreachable
                        # this command should not be necessary

fastdds shm clean # https://github.com/eProsima/Fast-DDS/issues/2003#issuecomment-1160245640
log ""

while true; do

    #ros2 param set /ann_controller genome_id test

    #ros2 param set /ann_controller update_genome_id True
    
    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"
    log "[DEBUG] Stop simulation"
    #ign service -s /world/arena/control --reqtype ignition.msgs.WorldControl --reptype ignition.msgs.Boolean --timeout 5000 --req 'pause: true'
    ign topic -t /pause_resume/control -m ignition.msgs.Boolean -p "data: true"

    # COUNTER=$((COUNTER + 1))
    # log "call $COUNTER" 
    # log "[DEBUG] Respawn turtlebot"
    # # ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'turtlebot4' position: { x: 1, y: 1, z: 1 } orientation: { x: 0, y: 0, z: 0, w: 1}" --timeout 2000
    # ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req 'name: "turtlebot4" position: { x: 1.0, y: 1.0, z: 0.2 } orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }' --timeout 5000
    # sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER" 
    log "[DEBUG] Respawn object1"
    #ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object1' position: { x: -1.5, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000
    ign topic -t /object1/pose_cmd -m ignition.msgs.Pose -p "position: {x: -1.5, y: 3.0, z: 2.0}, orientation: {w: 1.0, x: 0.0, y: 0.0, z: 0.0}"
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER" 
    log "[DEBUG] Respawn object2"
    #ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object2' position: { x: -1.2, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000   
    ign topic -t /object2/pose_cmd -m ignition.msgs.Pose -p "position: {x: -1.2, y: 3.0, z: 2.0}, orientation: {w: 1.0, x: 0.0, y: 0.0, z: 0.0}"
    sleep 1
    
    COUNTER=$((COUNTER + 1))
[... 21 lines cut ...]
    log "call $COUNTER"   
    log "[DEBUG] Respawn object6"
    #ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object6' position: { x: 1.1, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000 
    ign topic -t /object6/pose_cmd -m ignition.msgs.Pose -p "position: {x: 1.1, y: 3.0, z: 2}, orientation: {w: 1.0, x: 0.0, y: 0.0, z: 0.0}"
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"
    log "[DEBUG] Respawn object7"
    #ign service -s /world/arena/set_pose --reqtype ignition.msgs.Pose --reptype ignition.msgs.Boolean --req "name: 'object7' position: { x: 1.3, y: 3, z: 2} orientation: { x: 0.0, y: 0.0, z: 0.0, w: 1.0 }" --timeout 5000
    ign topic -t /object7/pose_cmd -m ignition.msgs.Pose -p "position: {x: 1.3, y: 3.0, z: 2}, orientation: {w: 1.0, x: 0.0, y: 0.0, z: 0.0}"
    sleep 1

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"
    log "[DEBUG] Start simulation"
    #ign service -s /world/arena/control --reqtype ignition.msgs.WorldControl --reptype ignition.msgs.Boolean --timeout 5000 --req 'pause: false'
    ign topic -t /pause_resume/control -m ignition.msgs.Boolean -p "data: false"

    sleep 10

done
```

The problem is that also in this case the simulation stops working after a while (I don't have detailed results yet), so I'm start thinking if the problem resides in something that is not related to the service call.

caguero, 2025-02-03:
Thanks for all the work on this issue @paoloelle . Do you think you could try to reproduce this issue using an example that only uses `gz-transport`? We'll be more confident debugging the issue isolating the problem a bit more.

paoloelle, 2025-02-04:
@caguero thanks for the reply. As mentioned here

> [@caguero](https://github.com/caguero) just a follow-up. I also tried the setup where I move only the position of the objects without launching anything related to the Turtlebot 4, and the problem [...]

I just moved the position of the objects using service calls and without spawning any robot. If there is other test that I can run please let me know.

caguero, 2025-02-04:
What I mean is an example that doesn't require to load Gazebo.

paoloelle, 2025-02-05:
OK, I got it. I will work on it. Anyway, I tried to simulate using a new version of these [two plugins](https://github.com/paoloelle/world_control_plugin/tree/8632447677b6c7d9eafdfdd624ba7be648a23cb0) that I implemented. The one named PauseResume stops the simulation and allows it to run for specified time intervals (without using any information from outside the simulation). The same applies to the MoveModel plugin, where the position of the objects is modified every time the simulation is stopped. This is a modified version of the same plugins mentioned[here](https://github.com/gazebosim/gz-transport/issues/564#issuecomment-2615208515).

So, my guess is that once I start using some form of communication (e.g., topic publishing or a service call), something breaks at some point.

paoloelle, 2025-02-06:
@caguero could you please provide me a link with an example or something that can help me implement this test?

caguero, 2025-02-06:
> [@caguero](https://github.com/caguero) could you please provide me a link with an example or something that can help me implement this test?

Probably the easiest thing is to look at the examples directory: https://github.com/gazebosim/gz-transport/tree/gz-transport14/example

paoloelle, 2025-02-07:
> > [@caguero](https://github.com/caguero) could you please provide me a link with an example or something that can help me implement this test?
> 
> Probably the easiest thing is to look at the examples directory: https://github.com/gazebosim/gz-transport/tree/gz-transport14/example

I just implemented a modified version of the`requester` node by adding a `while` loop that call the service offered by the `responser` every seconds. ```cpp
/*
 * Copyright (C) 2014 Open Source Robotics Foundation
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *     http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 *
*/

#include <iostream>
#include <ignition/msgs.hh>
#include <gz/transport.hh>

//////////////////////////////////////////////////
int main(int argc, char **argv)
{
  // Create a transport node.
  gz::transport::Node node;

  // Prepare the input parameters.
  gz::msgs::StringMsg req;
  req.set_data("HELLO");

  gz::msgs::StringMsg rep;
  bool result;
  unsigned int timeout = 5000;

  int requestCount = 0;

  while (true)
  {
    // Request the "/echo" service.
    bool executed = node.Request("/echo", req, timeout, rep, result);

    if (executed)
    {
      if (result)
        std::cout << "Response: [" << rep.data() << "]" << std::endl;
      else
        std::cout << "Service call failed" << std::endl;
    }
    else
      std::cerr << "Service call timed out" << std::endl;

    // Increment the request counter.
    requestCount++;
    std::cout << "Number of requests made: " << requestCount << std::endl;

    // Add a small delay to avoid spamming the service.
    std::this_thread::sleep_for(std::chrono::milliseconds(1000));
  }
}
```

After >14000 service requests I don't see any errors in genera nor errors like `NodeShared::RecvSrvRequest() error sending response: Host unreachable`.

Could there be a performance difference between calling a service from the terminal using a bash file and calling it via API (as done in `requester`)?

paoloelle, 2025-02-07:
Follow up of the previous (answer](https://github.com/gazebosim/gz-transport/issues/564#issuecomment-2642594163). If I try to call the service from the following `bash` script
```bash
#!/bin/bash

INTERVAL=1

START_TIME=$(date +%s)

log() {
    CURRENT_TIME=$(date +%s)
    ELAPSED=$((CURRENT_TIME - START_TIME))
    HOURS=$((ELAPSED / 3600))
    MINUTES=$(((ELAPSED % 3600) / 60))
    SECONDS=$((ELAPSED % 60))
    printf "[$(date +"%H:%M:%S")][%02d:%02d:%02d] %s\n" "$HOURS" "$MINUTES" "$SECONDS" "$1"
}

COUNTER=0 # counter for the call to ign services

while true; do

    COUNTER=$((COUNTER + 1))
    log "call $COUNTER"

    # Call ign service
    ign service -s /echo --reqtype ignition.msgs.StringMsg --reptype ignition.msgs.StringMsg --req 'data: "Hello from CLI"' --timeout 5000
    
    sleep $INTERVAL
done```

I can notice that the `Service call timed out` (and the message `NodeShared::RecvSrvRequest() error sending response: Host unreachable` on the `responder` terminal) message is becoming quite frequent after ~2500 service call.

caguero, 2025-02-14:
Testing it right now...

caguero, 2025-02-14:
I was able to reproduce it on Gazebo Ionic. After a few hundreds I start seeing from time to time:
```
NodeShared::RecvSrvRequest() error sending response: Host unreachable
```

paoloelle, 2025-02-17:
Could be somehow related to #243? More specifically to your [answer](https://github.com/gazebosim/gz-transport/issues/243#issuecomment-840069960) @caguero?. In the documentation the description related to the parameters `GZ_TRANSPORT_RCVHWM` states:
_Description: Specifies the capacity of the buffer (High Water Mark) that stores incoming Gazebo Transport messages. Note that this is a global queue shared by all subscribers within the same process. A value of 0 means "infinite" capacity. As you can guess, there's no such thing as an infinite buffer, so your buffer will grow until you run out of memory (and probably crash). If your buffer reaches the maximum capacity data will be dropped._
and for `GZ_TRANSPORT_SNDHWM`:
_Description: Specifies the capacity of the buffer (High Water Mark) that stores outgoing Gazebo Transport messages. Note that this is a global queue shared by all publishers within the same process. A value of 0 means "infinite" capacity. As you can guess, there's no such thing as an infinite buffer, so your buffer will grow until you run out of memory (and probably crash). If your buffer reaches the maximum capacity data will be dropped._

Might be that after too many service call there is some buffer that is full and drops data?

paoloelle, 2025-02-17:
Anyway is interesting the fact that if I run [this code](https://github.com/gazebosim/gz-transport/issues/564#issuecomment-2642594163), or a slightly modified version of it, from the terminal everything works fine. If I execute the same code from a `bash` file or from a python file using [`subrpocess.run()`] (https://docs.python.org/3/library/subprocess.html#using-the-subprocess-module) I get the error `NodeShared::RecvSrvRequest() error sending response: Host unreachable`. Don't know if this can help in someway.

paoloelle, 2025-02-18:
The issue could be in the creation of the communication node when a service is called. In the sense that, in this [file](https://github.com/gazebosim/gz-transport/issues/564#issuecomment-2642594163), I create the node only once and then iterate the service calls within the while loop, whereas in other cases, by using [this script](https://github.com/gazebosim/gz-transport/issues/564#issuecomment-2643053308), I create the communication node every time. Could it be something reasonable?

ChinmayMundane, 2025-02-19:
@paoloelle  I'm having somewhat same issue. I have two service which resets the gazebo world as well as resets the pose of my vehicle. I am accessing this service repeatedly for my RL thing. However after some episodes the gazebo simulation stops/ breaks abruptly. 
I used to do the same in gazebo classic (repeatedly calling reset_world service but this never happened). Have you found the reason causing this?

paoloelle, 2025-02-21:
@ChinmayMundane my problem is that after a certain point, the message `NodeShared::RecvSrvRequest() error sending response: Host unreachable` starts appearing more frequently, causing multiple service calls to fail. However, my simulation does not crash as you described.

paoloelle, 2025-03-14:
> The issue could be in the creation of the communication node when a service is called. In the sense that, in this [...]
Just a quick update: I tested it multiple times by running evolutionary algorithm and calling a custom service >1000 times and simulation doesn't  crash. I believe that the problem is related to the creation of a Node at each service call from the CLI.