gazebosim/gz-transport #592: Add support for removing a single subscription handler

iche033, 2025-03-21:
## Desired behavior
Currently there is one `bool Node::Unsusbcribe(const std::string &_topic)` API which removes all subscription handlers for that topic in the Node. The feature request is to have a variant of the `Unsubscribe` API that supports removing just one specific subscription handler from the node. 

As an example, the current usage of Subscribe / Unsubscribe and their behavior is as shown below:

```cpp
// create one subscription
bool success1 = node.Subscribe(myTopic, callback1, opts);
// create another subscription
bool success2 = node.Subscribe(myTopic, callback2, opts);
// unsubscribes both callback1 and callback2
node.Unsubscribe(topic);
```

Ideally we can have something like:

```cpp
// create one subscription
Subscriber sub1 = node.Subscribe(myTopic, callback1, opts);
// create another subscription
Subscriber sub2 = node.Subscribe(myTopic, callback2, opts);
// unsubscribes just sub1 
node.Unsubscribe(sub1);
// alternative API - unsubscribes just sub2 
sub2.Unsubscribe();
```

The new APIs above is just for illustrative purposes. It is however a breaking change since it changes the return type from bool to a custom struct / class. 

To minimize breaking changes, another option would be to add overloaded `Subscribe(myTopic, callback1, opts, sub1)` API in which you can pass a reference for it to be set by the function.

## Alternatives considered
See above for alternative API suggestions

## Implementation suggestion
For Unsubscribing a single subscription from node, we'll need a function similar to `Node::Unsubscribe`. The difference being:

* Support removing handlers by a unique id
https://github.com/gazebosim/gz-transport/blob/3caed76e3b61b8e206df79d2efe493dd894f7670/src/Node.cc#L615
* Only notify publishers that the node is no longer interested in receiving published messages if no more handlers exists: https://github.com/gazebosim/gz-transport/blob/3caed76e3b61b8e206df79d2efe493dd894f7670/src/Node.cc#L644

caguero, 2025-03-26:
What about something like this:

```
Subscriber sub1 = node.Subscribe(myTopic, callback1, opts);
```
And then, no explicit `Unsubscribe()`, you just unsubscribe when `sub1` goes out of scope and it's destroyed.

caguero, 2025-03-26:
`sub1 = node.CreateSubscriber(topic, callback, opts)` looks very reasonable to me.

Another question, what should be the expected behavior if somebody calls `node.Unsubscribe(topic)` after having subscribed to `topic` with a `CreateSubscriber(topic,...)`. I can see two options:
  1. Nothings happens. The only way to unsubscribe to `topic` is destroying `sub1`.
  2. Unsubscribe from `topic`. Then, even if `sub1` is still in scope, no callbacks will be executed.

PR #608 (2025-05-05): Adds new Node::CreateSubscriber API

# 🎉 New feature

Closes #592

## Summary

Adds a new API in Node class, namely `CreateSubscriber`. This function returns a `Node::Subscriber` object that allows the user to manage the lifetime of the subscription handler. When the Node::Subscriber is deleted (e.g. goes out of scope), it automatically removes the subscription handler from node. In comparison to the existing `Node::Unsubscribe` function that removes *all* subscription handlers for a topic, the advantage of using this new API is that users can now remove just one single subscription handler wi [... cut]