gazebosim/gz-transport #593: Compatibility issues with protobuf v30 (cpp 6.30.0)

oysstu, 2025-03-25:
## Environment
- Main branch
- Protobuf v30

## Description
There are compatibility issues with protobuf v30 (cpp 6.30.0) and later due to protobuf moving from string to string_view in some of the API. First build error:

## Output
```
gz-transport/include/gz/transport/ReqHandler.hh:385:43: error: could not convert ‘((google::protobuf::Message*)((const gz::transport::v15::ReqHandler<google::protobuf::Message, google::protobuf::Message>*)this)->gz::transport::v15::ReqHandler<google::protobuf::Message, google::protobuf::Message>::reqMsg)->google::protobuf::Message::google::protobuf::MessageLite.google::protobuf::MessageLite::GetTypeName()’ from ‘google::protobuf::internal::GetTypeNameReturnType’ {aka ‘std::basic_string_view<char>’} to ‘std::string’ {aka ‘std::__cxx11::basic_string<char>’}
  385 |           return this->reqMsg->GetTypeName();
      |                  ~~~~~~~~~~~~~~~~~~~~~~~~~^~
      |                                           |
      |                                           google::protobuf::internal::GetTypeNameReturnType {aka std::basic_string_view<char>}
```

PaoloReyes, 2025-03-28:
Same problem here. Did you find a solution?

oysstu, 2025-03-28:
The solution is similar to what I did here: https://github.com/gazebosim/gz-msgs/pull/499

I.e., go through the build errors one-by-one and either
- Change gz functions to take string_view instead of const string&.
- Convert string_view protobuf returns to strings where a copy is necessary
- Use stringstream or another alternative for string concatenation

I haven't had time to do this for gz-transports yet.

PaoloReyes, 2025-03-28:
Thanks for the reference! I’ll go through the build errors and apply the necessary fixes. I'll submit a PR if that helps!!!

oysstu, 2025-05-18:
Fixed in https://github.com/gazebosim/gz-transport/pull/615