gazebosim/gz-transport #602: SEGV during parsing of a discovery message

spark404, 2025-04-09:
## Environment
* OS Version: MacOS 15.4 (24E248)
* Source or binary build?
Homebrew installed gz-transport14 (osrf/simulation/gz-transport14: stable 14.0.1, HEAD) and
Source tag gz-transport14_14.0.1

## Description
* Expected behavior: 
  * Passing unittest UNIT_Discovery_TEST
  
* Actual behavior: 
 ```
/Users/xxx/Projects/gz-transport/cmake-build-debug/bin/UNIT_Discovery_TEST --gtest_filter=* --gtest_color=no
Testing started at 16:54 ...
Running main() from /Users/htrippaers/Projects/Personal/gz-transport/test/gtest_vendor/src/gtest_main.cc
Process finished with exit code 139 (interrupted by signal 11:SIGSEGV)
```

## Steps to reproduce
1. brew install gz-ionic
2. clone gz-transport repo
3. `cmake -DCMAKE_BUILD_TYPE=Debug  -G Ninja -S /Users/xxx/Projects/Personal/gz-transport -B /Users/xxx/Projects/Personal/gz-transport/cmake-build-debug
4. ninja build
5. `/Users/xxx/Projects/Personal/gz-transport/cmake-build-debug/bin/UNIT_Discovery_TEST`

I've been able to reproduce this error on two different Mac both with the latest update and a fresh install of gz-ionic from homebrew

osrf/simulation/gz-msgs11: stable 11.0.2, HEAD
protobuf: stable 29.3 (bottled)

## Output
```
Running main() from /Users/htrippaers/Projects/Personal/gz-transport/test/gtest_vendor/src/gtest_main.cc
[==========] Running 11 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 11 tests from DiscoveryTest
[ RUN      ] DiscoveryTest.TestBasicAPI
[       OK ] DiscoveryTest.TestBasicAPI (1006 ms)
[ RUN      ] DiscoveryTest.WithoutCallingStart
[       OK ] DiscoveryTest.WithoutCallingStart (1013 ms)
[ RUN      ] DiscoveryTest.TestAdvertiseNoResponse
zsh: segmentation fault  
```

[image]

spark404, 2025-04-16:
This seems to be related to an issue with protobuf.

No idea what exactly but after some trial and error rebuilding protobuf from source seems to fix my issue.

```
brew reinstall protobuf --build-from-source --debug-symbols
```

closed 2025-04-16 by spark404

scpeters, 2025-07-24:
I see that you are building gz-transport with `-DCMAKE_BUILD_TYPE=Debug`. I wouldn't expect that protobuf would need to be recompiled, but I'm guessing that is related. In general, I wouldn't expect macOS users to need to recompile protobuf from source with debug symbols if they are not also using `-DCMAKE_BUILD_TYPE=Debug`