gazebosim/gz-transport #701: Occasional crash on startup in discovery

iche033, 2025-08-28:
## Environment
* OS Version: ubuntu 20.04
* Source or binary build?
  source build, `gz-transport15` branch
  

## Description
* Expected behavior: gazebo + gz-transport starts up fine without crashing
* Actual behavior: Once in a while I get a crash with a backtrace pointing to gz-transport

## Steps to reproduce

1. launch gz-sim with any world

## Output

Crash happens occasionally, and gz-sim spits out the backtrace below that points to the use of `std::getenv`

<details><summary>backtrace</summary>

```
#7    Source "../../../../../src/libstdc++-v3/src/c++11/thread.cc", line 104, in execute_native_thread_routine [0x7f1516a76db3]
#6    Source "/home/iche/code/gz_j_ws_noble/src/gz-transport/src/Discovery.hh", line 1032, in RecvMessages [0x7f15177bf1a1]
       1030:           if (pollSockets(this->sockets, timeout))
       1031:           {
      >1032:             this->RecvDiscoveryUpdate();
       1033: 
       1034:             if (this->verbose)
       1035:               this->PrintCurrentState();
#5    Source "/home/iche/code/gz_j_ws_noble/src/gz-transport/src/Discovery.hh", line 1100, in __builtin_strlen [0x7f15177bf067]
       1097:                 << srcAddr << ": " << srcPort << std::endl;
       1098:             }
       1099: 
      >1100:             this->DispatchDiscoveryMsg(srcAddr, rcvStr + sizeof(len), len);
       1101:           }
       1102:         }
       1103:         else if (received < 0)
#4  | Source "/home/iche/code/gz_j_ws_noble/src/gz-transport/src/Discovery.hh", line 1266, in operator()
    |  1264:           {
    |  1265:             if (subscribersReqCb)
    | >1266:               subscribersReqCb();
    |  1267: 
    |  1268:             break;
      Source "/usr/include/c++/13/bits/std_function.h", line 591, in DispatchDiscoveryMsg [0x7f15177be7cc]
        588:       {
        589:    if (_M_empty())
        590:      __throw_bad_function_call();
      > 591:    return _M_invoker(_M_functor, std::forward<_ArgTypes>(__args)...);
        592:       }
        593: 
        594: #if __cpp_rtti
#3  | Source "/home/iche/code/gz_j_ws_noble/src/gz-transport/src/NodeShared.cc", line 1390, in SendSubscribersRep
    |  1388:   // Reply to the SUBSCRIBERS_REQ with multiple SUBSCRIBERS_REP.
    |  1389:   for (auto const &publisher : pubs)
    | >1390:     this->dataPtr->msgDiscovery->SendSubscribersRep(publisher);
    |  1391: }
      Source "/home/iche/code/gz_j_ws_noble/src/gz-transport/src/Discovery.hh", line 565, in OnSubscribers [0x7f15177a8462]
        562:       /// \param[in] _pub Information to send.
        563:       public: void SendSubscribersRep(const MessagePublisher &_pub) const
        564:       {
      > 565:         this->SendMsg(
        566:           DestinationType::ALL, msgs::Discovery::SUBSCRIBERS_REP, _pub);
        567:       }
#2  | Source "/home/iche/code/gz_j_ws_noble/src/gz-transport/src/Discovery.hh", line 1382, in Version
    |  1380:       {
    |  1381:         gz::msgs::Discovery discoveryMsg;
    | >1382:         discoveryMsg.set_version(this->Version());
    |  1383:         discoveryMsg.set_type(_type);
    |  1384:         discoveryMsg.set_process_uuid(this->pUuid);
      Source "/home/iche/code/gz_j_ws_noble/src/gz-transport/src/Discovery.hh", line 1574, in SendMsg<gz::transport::v15::MessagePublisher> [0x7f151778cd55]
       1571:         static std::string gzStats;
       1572:         static int topicStats;
       1573: 
      >1574:         if (env("GZ_TRANSPORT_TOPIC_STATISTICS", gzStats) && !gzStats.empty())
       1575:         {
       1576:           topicStats = (gzStats == "1");
       1577:         }
#1    Source "/home/iche/code/gz_j_ws_noble/src/gz-transport/src/Helpers.cc", line 41, in env [0x7f1517777144]
         38:     size_t sz = 0;
         39:     _dupenv_s(&v, &sz, _name.c_str());
         40: #else
      >  41:     v = std::getenv(_name.c_str());
         42: #endif
         43:     if (v)
         44:     {
#0    Source "./stdlib/getenv.c", line 31, in getenv [0x7f15167c0806]
Segmentation fault (Address not mapped to object [0x5570fb534])
Segmentation fault (core dumped)
``

</details>

iche033, 2025-08-29:
Another user ran into this crash during the tutorial party: https://github.com/gazebosim/gazebo_test_cases/issues/2154#issuecomment-3234876388

arjo129, 2025-09-19:
I can replicate it as well.

arjo129, 2025-09-19:
This may be a result of a `static` being used here.

https://github.com/gazebosim/gz-transport/blob/93b811cf2e8fd2743a433c92730dd1396ecdb3f9/src/Discovery.hh#L1571

closed 2025-12-17 by caguero